Tircown/ERCF-easy-brd #1: Voltage

cdkonecny, 2021-07-30:
Should voltage be stepped down to 12v for input on this board?
I have references to overheating at 24v but information is vague and unclear whether it applies to this version.

Tircown, 2021-07-30:
- At 12V the ERCF-EASY-BRD works fine, without issues. 
- At 24V you will need a good heatsink and to be sure that the servo motor doesn't block otherwise it draws more current and the voltage regulator goes into thermal protection. More test are still needed to know exactly when the heatsink is enough and when it's not.
- Another option is to deliver 24V to the board for the steppers motors and wire a buck converter in place of the L7805 to get 5V for the servo motor.

cdkonecny, 2021-07-30:
Very helpful.   Thank you

closed 2021-07-30 by cdkonecny