Tircown/ERCF-easy-brd #7: Cannot configure toolhead pin on board

Makarintosh, 2022-01-23:
I've built the ERCF using an Easy BRD 1.1 but I cannot find the pin that uses the "Extra" pin header on the board. The config shows the pin for SKR 1.4 but documents say I can use the "Extra" header.

Makarintosh, 2022-01-23:
```
## Enraged Rabbit : Carrot Feeder V1.1 hardware config file
# Values are an example for voron 2.4 with 2 SKR1.4

[mcu ercf]
serial: /dev/serial/by-id/usb-Klipper_samd21g18a_0D55BD144E34555020312E3132040DFF-if00

# Carrot Feeder 5mm D-cut shaft
# Example for an SKR 1.4 Board (E1 on the XY mcu)
[manual_stepper gear_stepper]
step_pin: ercf:PA4
dir_pin: ercf:PA10
enable_pin: !ercf:PA2
rotation_distance: 22.6789511	#Bondtech 5mm Drive Gears
gear_ratio: 80:20
microsteps: 16
full_steps_per_rotation: 200	#200 for 1.8 degree, 400 for 0.9 degree
velocity: 35
accel: 150
#Right now no pin is used for the endstop, but we need to define one for klipper. So just use a random, not used pin
endstop_pin: ^ercf:PA7

[tmc2209 manual_stepper gear_stepper]
# Adapt accordingly to your setup and desires
# The default values are tested with the BOM NEMA14 motor
# Please adapt those values to the motor you are using
# Example : for NEMA17 motors, you'll usually set the stealthchop_threshold to 0
# and use higher current
uart_pin: ercf:PA8
uart_address: 0
interpolate: True
run_current: 0.40
hold_current: 0.1
sense_resistor: 0.110
stealthchop_threshold: 500 

# Carrot Feeder selector
# Example for an SKR 1.4 Board (Z1 on the XY mcu)
[manual_stepper selector_stepper]
step_pin: ercf:PA9
dir_pin: ercf:PB8
enable_pin: !ercf:PA11
microsteps: 16
rotation_distance: 40
full_steps_per_rotation: 200	#200 for 1.8 degree, 400 for 0.9 degree
velocity: 200
accel: 600
endstop_pin: ^!ercf:PB9

[tmc2209 manual_stepper selector_stepper]
uart_pin: ercf:PA8
uart_address: 1
run_current: 0.55
interpolate: True
sense_resistor: 0.110
stealthchop_threshold: 5000

# Values are for the MG90S servo
[servo ercf_servo]
pin: ercf:PA5
maximum_servo_angle: 180
minimum_pulse_width: 0.00085
maximum_pulse_width: 0.00215

[duplicate_pin_override]
pins: ercf:PA6
# Put there the pin used by the encoder and the filament_motion_sensor
# It has to be the same pin for those 3

[filament_motion_sensor encoder_sensor]
switch_pin: ^ercf:PA6
pause_on_runout: False
detection_length: 4.0
extruder: extruder
# runout_gcode: ERCF_ENCODER_MOTION_ISSUE

[filament_switch_sensor toolhead_sensor]
pause_on_runout: False
switch_pin: ^P1.27
```

My current settings. I have tried cycling through different pin numbers but no luck.

Tircown, 2022-01-23:
1/ Change the gear_stepper's endstop_pin to an unused but existing pin on the Seeeduino XIAO, PA1 for instance.
2/ Put the jumper on the board between 1* and PA7
3/ The pin to set for the extra connector is now `^ercf:PA7`

Makarintosh, 2022-01-23:
Thanks for the help @Tircown. 

It didn't help however, Mainsail indicates that the toolhead_sensor is 'detected' but there is no filament at the sensor. I fed some filament in but nothing appears to change. Is there something else I can try?

[image]

Tircown, 2022-01-23:
The issue is now related to the toolhead sensor itself. 
This may be due to the wires, the strength of the magnet, etc.
Voron's discord server, #ercf_questions is a better place to get help.

closed 2022-01-23 by Tircown