Tircown/ERCF-easy-brd #19: Unbrick Seeduino

lettore, 2023-07-08:
As everyone knows it's easy to brick a Seeduino as the chip doesn't have a separate USB controller.
The 2 pins needed to reflash the bootloader are located on the back of the board, with this design they are unreachable and the only solution is to desolder the entire board or throw away all.
I'm wondering why the Seeduino is not fixed to the board through a removable socket, or at least provide two holes on the mainboard to reach that pads.
This is unbelievable.

Tircown, 2023-08-05:
The footprint has holes and the BOM contains optional 7P Header socket female (the male included with the Seeeduino XIAO), so you have the choice to solder directly with castellated pin or removable.
Also some vendors sales with this pins and a removable Xiao.

closed 2023-08-05 by Tircown

lettore, 2023-08-05:
As you say optional you don't understand that this is critical feature, to be able to reflash a bricked seeduino. Vendors will take this optional as an option ofcourse, why they should spend more for something that is optional for the designer. Anyway I bought a ERCF-Easy-brd can board that is based on RPI 2040 and works marvelous not like your design. I just wasted 30 Euro for a bricked board. Thank you