Tircown/ERCF-easy-brd #22: Need pinout for "Extra" port

Beta1492, 2024-07-16:
I'm trying to add a toolhead sensor to my ERCF using the "Extra" port on the Easy BRD, but I cannot find the pinout anywhere.
Any help would be appreciated!

Beta1492, 2024-07-16:
Found it! PA7
For anyone using the Easy BRD with Happy Hare, you need to use mmu:PA7 to call the pin.

closed 2024-07-16 by Beta1492

f00d4tehg0dz, 2024-08-14:
> Found it! PA7 For anyone using the Easy BRD with Happy Hare, you need to use mmu:PA7 to call the pin.
I'm having trouble getting a servo for the filament cutter to respond from PA7. I can make a diode LED turn on and off. But I can't get my servo to turn. How did you setup PA7 pin layout? 
For example, I have a DC 12v to a buck converter to 5v which runs directly to the servo, and the data/signal cable running to the PA7 1st pin out of 3.